JFC-Group/JF-Customisation #3: Failed to update configuration (checksum error)

KngKong, 2022-03-06:
I tried enabling FTP in my JioFiber following your steps,
While restoring it gives me checksum error even though i changed `config.checksum = "0"` in my decrypted file and continued.

My router description : 
    Model : JCO4032
    Firmware : SRCMTF1_JCO4032_R2.35

itsyourap, 2022-03-07:
The ```config.checksum = "0"``` was a bug (maybe intentional) and has been fixed in R2.35 firmware. 

FTP will not work after R2.35 firmwares.

closed 2022-03-07 by itsyourap

KngKong, 2022-03-13:
@itsyourap  Is there any other way to use the FTP.

KngKong, 2022-03-14:
Is there any other way to enable FTP in R2.35. Any other ways like downgrading the firmware or so...

itsyourap, 2022-03-25:
Yes! I have updated my repository! Please check