JFC-Group/JF-Customisation #5: How to connect to the router using telnet?

Black-bucky, 2022-04-25:
telnet: Unable to connect to remote host: Connection refused

I get this issue when I try to connect to the router 
using this command =  "telnet 192.168.29.1 23"

What should I do?

I have uploaded the file with the neccesary changes to my router (404)

Black-bucky, 2022-04-26:
> Please give a screenshot of the output you get after uploading the configuration ```.enc``` file

The router will restart in 120 seconds.

itsyourap, 2022-04-28:
Did you change the checksum too?

if yes, then don't. You **should not** change the checksum when trying to get root or enable telnet.

itsyourap, 2022-05-01:
Closing this issue due to inactivity. 

Reopen if needed.

Black-bucky, 2022-05-01:
> Did you change the checksum too?
> 
> if yes, then don't. You **should not** change the checksum when trying to get root or enable telnet.

Yes I did put checksum=0, what shoud I do else it would give checksum error?

itsyourap, 2022-05-03:
Yes exactly. We need the checksum error otherwise your router will reboot and changes will be reverted. 
Keep the checksum as it is and do not change it.

Black-bucky, 2022-05-03:
What should i do?

closed 2022-05-03 by Black-bucky