JFC-Group/JF-Customisation #10: JCO110 Key may be changed?

X3eRo0, 2022-06-12:
trying to decrypt the configuration file for JCO110 with firmware ARCTF1_JCO110_R1.148 leads to a bad decrypt.
Can you explain how did you extracted the encryption keys in the first place?
How did you reverse engineer the firmware? Thanks you are doing awesome work.

itsyourap, 2022-06-13:
> trying to decrypt the configuration file for JCO110 with firmware ARCTF1_JCO110_R1.148 leads to a bad decrypt.

Try this key : [server.key.zip](https://github.com/itsyourap/JioFiber-Home-Gateway/files/8887367/server.key.zip)! I just extracted it from the ```ARCTF1_JCO110_R1.148``` firmware!

> Can you explain how did you extracted the encryption keys in the first place?

Used [binwalk](https://github.com/ReFirmLabs/binwalk)

> Thanks you are doing awesome work.

Welcome

X3eRo0, 2022-06-13:
I wanna know more about the firmware. Is the firmware encrypted in sorts? 
Where did you get the firmware?
I wanna look for vulns in the actual webserver and shit. XD

X3eRo0, 2022-06-13:
I just extracted the ubifs from the firmware and I am looking at the webserver code. Its in Lua, I am so surprised XD