JFC-Group/JF-Customisation #11: JCOW404.key unable to decrypt my enc file

akiratoriyama, 2022-07-10:
OpenSSL gives the following error

*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
bad decrypt
20080000:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:providers\implementations\ciphers\ciphercommon_block.c:124:

itsyourap, 2022-07-11:
Can you tell me your firmware version?

Ghost-Rider10, 2022-07-12:
Same Problem But Terminal Give Different Output.

[image]

itsyourap, 2022-07-13:
Yes! The encryption password changed in the latest firmwares. Basically now the encryption key is specific for specific devices. The encryption key is a mixture of our router's serial number concatenated with the output of the command `/usr/bin/prolinecmd acspwd display`. Since the value of `acspwd` is unique for every device, I assume that there is no way to get root for now.

itsyourap, 2022-07-13:
Also the first five characters of `acspwd` matches the last five characters of your default Wi-Fi SSID name. I don't know about the origin of the next eleven characters. 

For example, if your default SSID is `JioFiber-XrHGa`, the first five characters of `acspwd` is `XrHGa`.

itsyourap, 2022-07-25:
**There is a solution to this**  🎊 

Use the WEB-UI to downgrade your router firmware to a version that used to work with this method.

closed 2022-07-25 by itsyourap

neelahagga, 2022-09-28:
> Use the WEB-UI to downgrade your router firmware to a version that used to work with this method.

Where can I find such a version for JCO4032?

itsyourap, 2022-10-01:
@neelahagga Idk- I didn't get a firmware for JCO4032 yet

itsyourap, 2023-01-03:
Use this to get Root: 
https://github.com/itsyourap/JioFiber-Home-Gateway/blob/master/Instructions/Alternate-Way-To-Get-Root-Access-JioFiber-ONT-Home-Gateway.md