JFC-Group/JF-Customisation #13: Decryption is not succesfull

Arkduke, 2022-10-24:
after using openssl to decrypt using the provided key file the file generated cosists of some chinese symbol and in the console its throwing message of bad decrypt. I am having router of firmware SRCMTF1_JCOW414_R2.41
[image]

itsyourap, 2022-10-25:
`R2.39` has introduced a device-specific way of encryption, which means that universal decryption isn't going to work. Currently, I don't have a way to get root access to firmware with versions `R2.39` and later.

Arkduke, 2022-10-25:
Okay then, i have to add another router🥲🥲.

itsyourap, 2023-01-03:
Please Check:
https://github.com/itsyourap/JioFiber-Home-Gateway/blob/master/Instructions/Alternate-Way-To-Get-Root-Access-JioFiber-ONT-Home-Gateway.md

closed 2023-01-04 by itsyourap