JFC-Group/JF-Customisation #19: not able to decrypt using the key from /flash/secure/key.txt

tanishqjasoria, 2023-01-04:
I got the key: RRIOTJI00000524XXXXXXXXXXXXXXXXXXXXXXXXXXXX (<serial_number><32 char string>)
then i used this command to decrypt the back up file: 
`openssl aes-128-cbc -d -pass file:key.txt -in RRIOTJI00000524_JCOW411.enc -out RRIOTJI00000524_JCOW411.txt `
`openssl aes-128-cbc -d -md md5 -pass file:key.txt -in RRIOTJI00000524_JCOW411.enc -out RRIOTJI00000524_JCOW411.txt `
had the same error
```*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
bad decrypt
4087D438AD7F0000:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:../providers/implementations/ciphers/ciphercommon_block.c:124:
```
`OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.2 15 Mar 2022)
`

tanishqjasoria, 2023-01-04:
apparently it was not needed - found the server.key in the firmware and it decrypted using that

itsyourap, 2023-01-05:
Can you tell me your Router Firmware version?

tanishqjasoria, 2023-01-05:
ARCNTF1_JCOW411_R2.21

itsyourap, 2023-01-05:
Completed via Telegram DMs

closed 2023-01-05 by itsyourap

Gangadhararao29, 2023-01-28:
I am also facing the same issue , I got the secret file but decreption is failing

itsyourap, 2023-01-28:
> I am also facing the same issue , I got the secret file but decreption is failing

@Gangadhararao29 Please tell me your firmware version

Gangadhararao29, 2023-01-29:
@itsyourap JCOW414 v2.49

itsyourap, 2023-01-29:
> @itsyourap JCOW414 v2.49

@Gangadhararao29 You can definitely downgrade your firmware from the WEB-UI and get root if you want.