JFC-Group/JF-Customisation #24: key for 407

xtendtech, 2023-02-08:
I got the firmware http://fota.slv.fxd.jiophone.net/ONT/Skyworth/JCOW411/SKYWTF1_JCOW407_R2.45.img for skyworth 407 which is latest fiber ont. Any way to get the key  to decrypt it?

AG4lyf, 2023-02-13:
JCOW414.key from https://github.com/itsyourap/JioFiber-Home-Gateway/blob/master/Old%20Encryption%20Keys%20(Might%20not%20work)/JCOW414.key works fine for me

AG4lyf, 2023-02-13:
although i got myself a telnet connection. and copied this 
https://drive.google.com/file/d/17MwdrZawn25MJ15iGqSNxNHcLAL2o8iL/view?usp=share_link

gh would not let me upload .key file

xtendtech, 2023-02-14:
I am getting this errir if i use that key  414  openssl aes-128-cbc -d -pass file:JCOW414.key  -in RNOOTLI40040484_JCOW407.enc -out output.txt
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
bad decrypt
140113619780928:error:06065064:digital envelope routines:EVP_DecryptFinal_ex:bad decrypt:../crypto/evp/evp_enc.c:615:

itsyourap, 2023-02-14:
@xtendtech You are using JCOW414 key for a JCOW407 ONT. You cannot do that.

Also if your router firmware is recent, then these keys won't work anymore.

AG4lyf, 2023-02-14:
have you tried the key i provided. 
for me 414 key worked somehow.

itsyourap, 2023-03-06:
Closing this issue due to inactivity.

closed 2023-03-06 by itsyourap