JFC-Group/JF-Customisation #25: Use gpon for our own use

adbishnoi507, 2023-02-13:
I have done the telnet part. And got access to it. 

I wanna use this as my own modem as my isp modem sucks 
My isp requires a vlan id of 528 and a pppoe username and password

I have tried configuring .enc and flashing it with no success. 

Any ways i can do this.

adbishnoi507, 2023-02-14:
How can i flash old firmware ? 
I have img file but what about the key?

xtendtech, 2023-02-14:
cant you find it in this github repo?

AG4lyf, 2023-02-14:
do you know the exact version ? 
we went to R1.12. and it is now bad cipher. 

```bad decrypt
74560100:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:providers\implementations\ciphers\ciphercommon_block.c:124:```

AG4lyf, 2023-02-14:
2.2 also having the same issue.

xtendtech, 2023-02-14:
i think it might be the key  which they posted might have changed.. check the way to get root and try to copy it from the router and then use it

closed 2023-03-06 by itsyourap