JFC-Group/JF-Customisation #27: getting error while decrypting

VIKRAMSSJK, 2023-03-17:
Microsoft Windows [Version 10.0.19045.2604]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\system32>cd C:\Users\CAPTAIN\Downloads

C:\Users\CAPTAIN\Downloads>openssl aes-128-cbc -d -pass file:key.txt -in CAPTAIN.enc -out output.txt
Can't open file key.txt
Error getting password
F0070000:error:80000002:system library:BIO_new_file:No such file or directory:crypto\bio\bss_file.c:67:calling fopen(key.txt, r)
F0070000:error:10000080:BIO routines:BIO_new_file:no such file:crypto\bio\bss_file.c:75:

itsyourap, 2023-03-17:
The error is clearly written : `key.txt` is not found in `C:\Users\CAPTAIN\Downloads`

itsyourap, 2023-03-22:
Closing due to inactivity