JFC-Group/JF-Customisation #30: trying port forwarding on jio fiber?

rakite28, 2023-05-31:
hey so i am using my orange pi board as a pihole and DOH server on my JIO Fiber connection , i also tried to make a vpn using PiVPN it worked and the vpn was made using OpenVPN, but it uses the routers public ip address which is supposed to be dynamic but  hasnt changed ever so that isnt the issue but the issue seems to be port forwarding as i cant reach my jf router via th e public address, is therr any way to actually enable port forwarding on the router if i enable the root user, just wanted to know before i go down that path, would appreciate any help and ideas otherwise too.

Otus9051, 2023-06-04:
Jio uses CGNAT for IPv4, while you may be successful in port-forwarding via IPv6, you cannot port-forward IPv4. I would suggest getting a cheap VPS somewhere and tunneling your services through there. I personally just setup WireGuard.

monojitmondal04, 2023-07-18:
Use tailscale its free

closed 2023-08-10 by itsyourap