JFC-Group/JF-Customisation #32: Backup decryption not working

mrishi, 2023-07-09:
- I was able to download the backup key using the method mentioned here: https://github.com/JFC-Group/JF-Customisation/blob/master/Instructions/Get-Any-File-From-JF-ONT-Home-Gateway.md

- However, using this command: openssl aes-128-cbc -d -pass file:<key.txt> -in <input.enc> -out <output.txt>
I got this error:
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
bad decrypt
009EDFFF01000000:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:providers/implementations/ciphers/ciphercommon_block.c:124:

Pls. help..

itsyourap, 2023-07-10:
The encryption keys have changed since then.
Use https://github.com/JFC-Group/JF-Customisation/blob/master/keyguesser.py

mrishi, 2023-07-10:
Thanks but same result. Btw, I am on firmware 2.29 on JCOW404 model.

Also, I noticed that the key dumped by old method was in the format: <Router Serial><DefaultSSID Suffix><some hex key>, which seems to be as per the permutations you are generating in keyguesser.py.

So, I tried to add the hex-key part of the key to your keys list in keyguesser.py, so that it could be used in the generated permutations, but that didn't help either.

closed 2024-04-24 by itsyourap