JFC-Group/JF-Customisation #33: Firmware URL changed

souravdas142, 2023-09-05:
Hi,

The firmware URL changed. currently with your method I am getting upto 2.49, but last night router upgraded to 2.52, and I didn't get the url for that firmware.

[image]

```bash
└──╼ $ wget --spider http://fota.slv.fxd.jiophone.net/ONT/Sercomm/JCOW414/SRCMTF1_JCOW414_R2.52.img
Spider mode enabled. Check if remote file exists.
--2023-09-05 08:34:42--  http://fota.slv.fxd.jiophone.net/ONT/Sercomm/JCOW414/SRCMTF1_JCOW414_R2.52.img
Resolving fota.slv.fxd.jiophone.net (fota.slv.fxd.jiophone.net)... 2405:200:1606:1731::312c:b0b9, 2405:200:1606:1731::312c:b0ba, 49.44.194.80, ...
Connecting to fota.slv.fxd.jiophone.net (fota.slv.fxd.jiophone.net)|2405:200:1606:1731::312c:b0b9|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

itsyourap, 2023-09-05:
Yes I know..
We also have not been able to track the latest upgrade URL since it happened without notice..

We are trying...

d5aqoep, 2023-09-23:
Is it possible to downgrade the firmware? People are saying they are unable to downgrade.

Derpitron, 2024-03-24:
Anyone know any method?

itsyourap, 2024-04-24:
The only way we know how to downgrade is to already have root access to the router.

Closing this issue now. If any breakthrough arrives, we will reopen this

closed 2024-04-24 by itsyourap

Naitik1208, 2025-06-26:
https://fota.slv.kai.jiophone.net/XPGNO/DIN460/S404W_102/yTPaHbfA4q.img i got a link for 3.10.2 can anyone fetch more links

> The only way we know how to downgrade is to already have root access to the router.
> 
> Closing this issue now. If any breakthrough arrives, we will reopen this

ullu1221, 2026-10-15:
> 7\. https://fota.slv.kai.jiophone.net/XPGNO/DIN460/S404W_102/yTPaHbfA4q.img i got a link for 3.10.2 can anyone fetch more links
>         > The only way we know how to downgrade is to already have root access to the router.
>         > Closing this issue now. If any breakthrough arrives, we will reopen this

link for 2.15 https://raw.githubusercontent.com/gpsarathy/JioFiber-Home-Gateway/master/Firmwares/JCOW414/2.15/SRCMTF1_JCOW414_R2.15.img
https://raw.githubusercontent.com/gpsarathy/JioFiber-Home-Gateway/master/Firmwares/JCOW414/2.15/SRCMTF1_JCOW414_R2.15.sig

princepathria, 2026-10-29:
Not able to downgrade JCOW414 to R2.15 from SRCMTF1_JCOW414_R2.57.
Error- Firmware file is invalid or corrupted.

imiakk, 2026-11-05:
same here dude