JFC-Group/JF-Customisation #34: Decryption not working: ARCNTF1_JCOW411_R2.29

lscambo13, 2023-09-11:
command used after generating key from _keyguesser.py_: 
`openssl aes-128-cbc -d -pass file:key-success.txt -in RAROTKH00053021_JCOW411.enc -out RAROTKH00053021_JCOW411.txt`

output:
```
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.

```

**Problem**: the decrypted txt file is still just a garbled text.

**Firmware Version**: ARCNTF1_JCOW411_R2.29

Internet suggested to use LibreSSL, but I couldn't get the windows version to work.
_I don't have a mac._

Vshindou, 2024-02-02:
open the keyguesser.py make the following changes 

1. Open your key.txt
2. Check the text after routerSSID in the key.txt.
3. Add these to the keyString variable as shown below

keyStrings = ["1n0NaZQnC9oxcfwf", "us4AQiJAgbj0Fmxq", "NTqK8Ps5iFke8zrp", "bfqerloC15y79WQZ",
              "9gNzEbuDjtyT9Pyc", "uuphsZuO92AZW5GJ", "qdySWmmvYKdBcO53", "Q7ODauKsxUAUtbR7",
              "Kohgiem4joochei3","XXXXXXXXXX"]

4. Here replace the X with your characters mine was 10 char long yours might be longer or shorter
5.  Save keyguesser.py and run it again
6. It gives that key depreciated error that can be ignored

itsyourap, 2024-04-24:
Closing due to no activity

closed 2024-04-24 by itsyourap