JFC-Group/JF-Customisation #45: Run script in startup

harsha-0110, 2024-05-17:
Is there a way to run shell scripts during the router boot up?

itsyourap, 2024-06-04:
Yes there is.

- Keep you script in `/flash2/pfrm2.0/etc/customInit`
- Remember to keep the file in LF
- `chmod +x /flash2/pfrm2.0/etc/customInit`
- Restart your router to test

**Don't Use Such Scripts that Might Take long to run! If it is a long running script, use nohup otherwise your router may never startup**

closed 2024-06-04 by itsyourap

xdevcoder1, 2025-04-07:
Hey did the above steps but my script does not after run reboot

i mounted the pfrm2.0 to /flash2/pfrm2.0 but it was read only
so just copy pfrm2.0 to /flash2/pfrm2.0 
This is my script and I named it customInit and its in LF and I made it executable with chmod but this does not run on after a reboot

````
#!/bin/sh

# Wait for the system to be ready

sleep 10

# Remove the telnet disable flag
rm /flash/telnetDisable
rm /flash/telnetDisableACS

# Start telnet and allow it through the firewall
/usr/sbin/telnetd
/pfrm2.0/bin/iptables -I fwInBypass -p tcp --dport 23 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT

# Set the root password
echo -e "rilpass\nrilpass" | passwd root

#Ftp Enable
vsftpd /flash/vsftpd.conf &
iptables -I fwInBypass -p tcp --dport 21 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT
````

itsyourap, 2025-04-08:
@xdevcoder1 Why would you mount `/pfrm2.0` or copy anything from it?

You just need to keep your script in `/flash2/pfrm2.0/etc/customInit`

Also what is your firmware version? The method is tested till R2.57 for now.
You can try using `nohup` to run the commands and check if they work

xdevcoder1, 2025-04-08:
oh i am new to these things so sorry if that was a stupid question to ask and It did because there is no pfrm2.0 dir so i just assumed i need to those things but never mined it work's now.
Thanks for answering question.