JFC-Group/JF-Customisation #49: Backup file decryption error

shamxl, 2024-06-20:
```~/JFC $ openssl aes-128-cbc -d -pass file:key.txt -in RKIOTFI20164045_JCOW407.enc -out output.txt```

*** WARNING : deprecated key derivation used.                                                   Using -iter or -pbkdf2 would be better.                                                         bad decrypt                                                                                     00858F6071000000:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:providers/implementations/ciphers/ciphercommon_block.c:107:

firmware: `SKYWTF1_JCOW407_R2.52.2`

shamxl, 2024-06-20:
this is my key.txt btw

`RKIOTFI20164045fKduM25u52Fmf7yk`

itsyourap, 2024-06-21:
The key.txt should contain the HEX key as output by the keyguesser.py

closed 2024-06-21 by itsyourap