JFC-Group/JF-Customisation #51: Failed to update router firmware

Ni8htOwl, 2024-06-30:
Hi, I'm trying to update my jiofiber router firmware from the last couple of days but everytime I'm getting an error saying firmware file is invalid or corrupted. Tried multiple update files but getting the same error while upgrading/downgrading. Is this a know issue or am I missing something? Please look into this issue. Thanks.
 
Here's my router information :
Model - Skyworth JCOW407
Current firmware version - SKYWTF1_JCOW407_R2.52.2

leitogs, 2024-07-01:
> Hi, I'm trying to update my jiofiber router firmware to the latest version(R2.9) but I'm getting an error saying firmware file is invalid or corrupted. Tried multiple update files but getting the same error. Is this a know issue or am I missing something? Please look into this issue. Thanks.
> 
> Here's my router information : Model - Skyworth JCOW407 Current firmware version - SKYWTF1_JCOW407_R2.52.2

I guess R2.52.2 is the latest. R2.9 firmware year is 2021, R2.52.2 shows 2023.
Open firmware img with 7zip check the date in /pfrm2.0/firmwareDate file.

Ni8htOwl, 2024-07-01:
Okay, but even if i try to downgrade the firmware, I'm getting the same error.

itsyourap, 2024-07-01:
> Okay, but even if i try to downgrade the firmware, I'm getting the same error.

@Ni8htOwl Yes! That is default behaviour. You cannot just "downgrade" your software from a "upgrade" page.

Ni8htOwl, 2024-07-01:
Yeah, makes sense.

Ni8htOwl, 2024-07-07:
Just wanted to check if it is working or not.

Ni8htOwl, 2024-07-07:
My router is already on latest firmware. That's why it is failing to update. Thanks for the response.

closed 2024-07-07 by Ni8htOwl

itsyourap, 2024-07-08:
We have a way to downgrade but if you already have root access.

Ni8htOwl, 2024-07-08:
> We have a way to downgrade but if you already have root access.

Ohh. I'm not looking to downgrade. I also don't have root access. But, thanks for the reply.

Naitik1208, 2025-05-04:
How am i supposed to downgrade i think i have root access as I followed all steps mentioned in guide
The output after uploading enc file was Changing password for root
New password: 
Retype password: 
Changing password for root
New password: 
Retype password: 
Content-Type: text/html; charset=UTF-8