JFC-Group/JF-Customisation #58: KeyGuesser

GOAT0P, 2024-09-17:
What to do if Keyguesser can't provide the correct key?

itsyourap, 2024-09-18:
Depends on the version of your router firmware. If it is somewhat around or more than R2.52, then keyguesser is the only option for now. Generally, it should work.

If it doesn't, check if you can get your router firmware img using the following [tutorial](https://github.com/JFC-Group/JF-Customisation/tree/main/Router%20Firmware). If you cannot, then just comment down your router firmware version here.

GOAT0P, 2024-09-18:
SRCMTF1_JCOW414_R2.52.1

There you go man. Appreciate the help.

itsyourap, 2024-09-18:
It's a JCOW414 with R2.52.1
`keyguesser.py` should work with this

GOAT0P, 2024-09-18:
Nah man, it's not working.

itsyourap, 2024-09-19:
You have Telegram? If yes, then message me at [itsyourap](https://t.me/itsyourap) with your encrypted config file and your default Router SSID.

itsyourap, 2024-09-20:
Closing since issue is resolved via Telegram

closed 2024-09-20 by itsyourap