JFC-Group/JF-Customisation #59: Still working?

Shinchan7222, 2024-10-08:
I want to open ftp ( to access usb files) and port forwarding ( for my camera)

Shinchan7222, 2024-10-08:
(main) $ openssl aes-128-cbc -d -kfile "RAROTBJ00039044_JCOW411.enc" -in "RAROTBJ00039044_JCOW411.enc" -out "RAROTBJ00039044_JCOW411.txt"
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
bad decrypt
8082F6F227790000:error:1C800064:Provider routines:ossl_cipher_unpadblock:bad decrypt:providers/implementations/ciphers/ciphercommon_block.c:124:
@Shinchan7222

Shinchan7222, 2024-10-08:
Trying Key : zuFbKywMhJjVEhk35GRAROTBJ00039044
Success : AD816E42C391E7199E92705583073189
@Shinchan7222 ➜ /workspaces/IG-Maker (main) $ aes-128-cbc -d -pass file:key.txt -in RAROTBJ00039044_JCOW411.enc -out output.txt
bash: aes-128-cbc: command not found
@Shinchan7222 ➜ /workspaces/IG-Maker (main) $ openssl aes-128-cbc -d -pass file:key.txt -in RAROTBJ00039044_JCOW411.enc -out output.txt
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.
@Shinchan7222 ➜ /workspaces/IG-Maker (main) $

itsyourap, 2024-10-09:
> @Shinchan7222 ➜ /workspaces/IG-Maker (main) $ openssl aes-128-cbc -d -pass file:key.txt -in RAROTBJ00039044_JCOW411.enc -out output.txt
> *** WARNING : deprecated key derivation used. Using -iter or -pbkdf2 would be better.

This means that the decryption is successful. Please check the folder. It will contain the decrypted config `RAROTBJ00039044_JCOW411.txt`.

Shinchan7222, 2024-10-09:
It's not Decrypted it's in unreadable formate

Shinchan7222, 2024-10-09:
[RAROTBJ00039044_JCOW411.zip](https://github.com/user-attachments/files/17306502/RAROTBJ00039044_JCOW411.zip)
inFileName = "RAROTBJ00039044_JCOW411.enc" # Full name of the encrypted config backup file                    #
outFileName = "RAROTBJ00039044_JCOW411.txt" # Name of output file if the decryption is successful             #
routerSerial = "RAROTBJ00039044" # Your Router's Serial Number                                                #
routerSsid = "5G"  # Default Router SSID without the 'JioFiber-' prefix 

[image]

Shinchan7222, 2024-10-09:
can you please decrypt it for me if your free

Shinchan7222, 2024-10-09:
[image]

itsyourap, 2024-10-10:
> Your Router's Serial Number # routerSsid = "5G" # Default Router SSID without the 'JioFiber-' prefix

Here you have to use the default SSID which is written on the back of your Router box.
Something like "JioFiber-Nxyao"

So `routerSsid = "Nxyao"`

Shinchan7222, 2024-10-10:
THANK YOU I WAS ABLE TO DECRYPT IT BUT NEW ISSUE CAME UP

[image]

closed 2024-10-10 by itsyourap

Shinchan7222, 2024-10-10:
[image]
worked thank you so much bro, is there anyway to open a port? thanks

itsyourap, 2024-10-10:
> Is there anyway to open a port?

`iptables -I fwInBypass -p tcp --dport <PORT-NUMBER> -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT`

Shinchan7222, 2024-10-10:
sorry i meant port forward

itsyourap, 2024-10-10:
> sorry i meant port forward

Port Forwarding IPv4 isnt possible afaik. 
For IPv6, refer to [this article](https://www.reddit.com/r/HomeNetworking/comments/14g4udp/how_to_host_servers_on_jio_fiber_home_network/)

Shinchan7222, 2024-10-10:
[image]
ahhhhhhhh

Shinchan7222, 2024-10-10:
and can you tell me how do i run an sh file
on boot