JFC-Group/JF-Customisation #63: Disable TR-069

IfranAli8K, 2024-10-12:
How to disable the TR-069? Is there any way to disable it? The old methods seems to show a checksum error while uploading the file with modified TR-069 stats.

itsyourap, 2024-10-13:
The old method still works as expected. You have to re-encrypt the config with the valid checksum.

You can also get root and change the config file directly at `/flash/teamf1.cfg.ascii`, `/flash/teamf1.cfg.ascii.bkp` and `/tmp/system.db` and restart your router.

IfranAli8K, 2024-10-13:
How do I re-encrypt the config with the valid checksum?

itsyourap, 2024-10-14:
> How do I re-encrypt the config with the valid checksum?

https://github.com/JFC-Group/JF-Customisation/blob/main/Instructions/Encrypt-Router-Configuration-File.md

closed 2024-10-14 by IfranAli8K