JFC-Group/JF-Customisation #64: Any way to install packages on usb pen drive

Shinchan7222, 2024-10-26:
Any way to install packages on external usb plugged in by us i tried installing ffmpeg today with opt but i got storage issue since theres only a few mb storage in jio fiber

itsyourap, 2024-10-31:
Since this is a linux system, format your pendrive as ext4 and you can run binaries on it.

Shinchan7222, 2024-10-31:
I was able to install it but i am getting not found even though file exists

Shinchan7222, 2024-10-31:
It's like opkg is not working with pendrive

itsyourap, 2024-10-31:
Is the filesystem ext4?

itsyourap, 2024-10-31:
Also, you can checkout [this script](https://github.com/JFC-Group/JF-Customisation/blob/main/Entware/entware-mips.sh). It maybe useful

Shinchan7222, 2024-10-31:
Yes it is ext4

Shinchan7222, 2024-10-31:
The issue is with opkg no matter what it installs on root not pendrive

itsyourap, 2024-10-31:
You can try creating a alike file strucutre on the pendrive and copy and chroot into busybox on the pendrive and install your packages there. Afaik people have succeded with this one

Shinchan7222, 2024-10-31:
I don't know much about chroot but i will try to get help from gpt and thanks for your response

itsyourap, 2024-11-08:
@Shinchan7222 Any updates?

itsyourap, 2024-11-14:
Closing due to inactivity.