JFC-Group/JF-Customisation #65: Unable to locate Auto Update File/ sh/ script

ZooRes, 2024-10-30:
Hello Friends,

I am looking for auto update file in jio router filesystem, however I don't have much knowledge so its looking dificult to locate one by one all files. If anybody with root access know the file location please help me to understand how its get auto update.

Thank you so much.

itsyourap, 2024-10-31:
There are a lot of places where it can auto update:-
- Direct shell access through TR069
- Update check though Lua
- firmUpdateSystemConfig table in system.db (although the download URLs are empty)
- Even the `ont_dc` executable log has something related to updates

We don't really have any evidence on which one of these are being used.

viraniac, 2024-11-03:
Update for Jio fiber router happens via tr069 if you login to the router just after update you should be able to find the ota url in the tr69 logs.

itsyourap, 2024-11-06:
Currently, removing or modifying the TR069 ACS Server URL breaks the internet connection of the router so we cannot just disable TR069. 

I am not sure but we can modify the  `UPGRADE_PROGRAM` in `environment` table in `/tmp/system.db` to maybe disable updates.

viraniac, 2024-11-06:
> I am not sure but we can modify the `UPGRADE_PROGRAM` in `environment` table in `/tmp/system.db` to maybe disable updates.

Thats should be easy to test. I have switched back to 2.52.1 by changing the boot slot and have set PRE_UPGRADE_SCRIPT to /bin/false. In theory, the device should now attempt to upgrade and report back failure. Upgrade is not being pushed to my device right now, atleast I can't see it in the logs yet. Will report back if anything changes.

viraniac, 2024-11-08:
My plan expired today and after recharge finally got the update pushed again to me. Maybe Jio was waiting to see if I will recharge or not.

Anyways, I had set `POST_UPGRADE_SCRIPT` upgrade script set to `/bin/false` as I noticed that it was this script that actually flashed the update. I am seeing all the update attempts now downloading the file, trying to flash it and then say the file is corrupt. So simplest way to stop upgrades is simply setting any of the PRE_UPGRADE_SCRIPT, UPGRADE_PROGRAM or POST_UPGRADE_SCRIPT to /bin/false.

Now changeing my slot to be back again on 2.57.

itsyourap, 2024-11-08:
We could also:-

- Take a look at the original args of firmd
- Create a shell script in `/flash2` that does the following:-
	- Logs the args provided
	- Makes copies of  `/tmp/tr69*` files in `/flash2`
	- Copies the firmware file provided as args to the script to somewhere in `/flash2`
- Make the script executable with `chmod`
- Modify `UPGRADE_PROGRAM `to the script location

closed 2024-11-08 by itsyourap

viraniac, 2024-11-08:
> We could also:-
> 
> * Take a look at the original args of firmd
> * Create a shell script in `/flash2` that does the following:-
>   
>   * Logs the args provided
>   * Makes copies of  `/tmp/tr69*` files in `/flash2`
>   * Copies the firmware file provided as args to the script to somewhere in `/flash2`
> * Make the script executable with `chmod`
> * Modify `UPGRADE_PROGRAM` to the script location
> 
> This will give us the firmware and also the persistent firmware link in the TR069 logs.

Also user will have to add a script that modifies UPGRADE_PROGRAM in system.db on every boot as it gets overwritten. Its not a problem though as there is a way to do that as well. Anyone who will look into init script can easily find it. Best part is it persists across factory reset as well and I am using it to set my default configurations for jio router on boot that were getting changed otherwise. I am not going to say out loud how to achieve this, but as I said its pretty easy to figure out for anyone who goes through the init scripts.

itsyourap, 2024-11-08:
> Also user will have to add a script that modifies UPGRADE_PROGRAM in system.db on every boot as it gets overwritten.

#45

itsyourap, 2024-11-08:
> Yes, I am also using the same, though I didn't knew until after I posted my comment that you have found this as well. I found it during my own investigationss. One thing to note about it is it persists across factory resets. So it, in a way, is pretty bad as anyone can hack into this pathetic firmware and then can monitor or modify traffic flowing through it without user having any knowledge about it. And even if user resets the router, the hack will persist.

Yes it will persists unless Jio changes them with their firmware updates, thus this issue.

itsyourap, 2024-11-08:
@viraniac You can take a look at #67. Your suggestions are welcome.