JFC-Group/JF-Customisation #68: Old firmware SRCMTFI_JCO4032_R2.15 - Broken Links in Guide & Issues Accessing Key.txt with New Guide

PotatoPC44, 2024-11-13:
First of all, amazing work on your JioVoice research and all the related guides here on GitHub! I’ve gone through everything, and it’s been incredibly helpful.

I’m working with an old Jio router that I had lying around, so the firmware is really outdated (SRCMTFI_JCO4032_R2.15). I’m currently using a newer Jio router where these steps don’t work, as Jio patched the exploit a few weeks ago. Since this older firmware should still be vulnerable, I’m trying to gain root access on it to potentially use it instead of the new router (pending Jio’s approval).

I followed your older guide, but many critical links are broken now, including this one for Encryption Keys: https://github.com/JFC-Group/JF-Customisation/tree/master/EncryptionKeys/.

Without access to the older guide links, I tried the newer guide, including retrieving Key.txt with a POST request, but I couldn’t get it to generate any file on my FAT32-formatted USB drive (with a simple name that I used correctly in the request). I attempted multiple times with different cookie IDs and configurations but had no luck.

Any advice on restoring access to those old links or solving the Key.txt issue would be invaluable. Thanks again for all the hard work and research!

itsyourap, 2024-11-13:
Are you using the latest firmware (R2.57 or higher)?
If yes, then you don't need any keys....

You should use the keyguesser.

Also if you are using R2.57.1 and you cannot see the Backup and Restore page in your Router Admin Panel. you should follow [this](https://broadband.forum/threads/jiofiber-router-jco4032-settings-import-export.231300/post-1672959)

itsyourap, 2024-11-13:
However if you are looking for the default keys, check #52

PotatoPC44, 2024-11-13:
I’m using “R 2.15” not R2.57.1. It is firmware from 2021 this router has been lying with me for years, Jio never came to collect after i terminated the connection , so I need access to old guide 😅 but links are broken and new guide are not working 😅.

PotatoPC44, 2024-11-13:
I tried both 414 and 4032 keys but getting error "error reading input file". I will try POST request method again with some other pendrive.

itsyourap, 2024-11-14:
> I used the hex output of keyguesser.py and saved it into a key.txt file then used that file to encrypt again but got an error
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.

That isn't an error. Its just a warning.

> what else can i do now i have root access?

There isn't an answer to this question. If you don't know what to do with root access, then probably you shouldn't even bother to get the root access.