JFC-Group/JF-Customisation #73: 401 Unauthorized

adithyapi4, 2024-11-25:
I am getting the "401 Unauthorized" Error when I try to restore from the modified backup file for gaining root access. I reset the router and tried again with no change. Kindly post a fix to this issue. Thanks in advance.
[image]

itsyourap, 2024-11-25:
Can you tell me what steps have you exactly followed after decrypting the backup?

adithyapi4, 2024-11-25:
I was able to gain root access before the firmware was upgraded to SRCMTF1_JCO4032_R2.57

itsyourap, 2024-11-25:
Did you create the backup before upgrading or did you do it after upgrade?

adithyapi4, 2024-11-25:
First I tried using a backup file that I had created before upgrading and got the 401 Unauthorized error. So I tried again by downloading a new backup file and then restoring from it but got the same error

adithyapi4, 2024-11-26:
I am getting the same error even if I try to restore from the unmodified backup file after resetting the router to its factory defaults

itsyourap, 2024-11-26:
Thats unexpected. I think Jio messed something up (though it is very less likely since we have had no problems on R2.57).

PotatoPC44, 2024-12-18:
This method worked for my JCOW414 on 2.57 and just a another tip. I also got 401 error but mine was fixed my doing the steps under 2 minutes of logging in the router because router locks itself after 2-3 minutes.

adithyapi4, 2024-12-19:
Tried it within 1 minute. Still the same issue

itsyourap, 2024-12-19:
@adithyapi4 What do you have in key.txt?

It should have the hex key and not the regular key

itsyourap, 2024-12-19:
Yes the "Success: XXXX" is indeed the key required to decrypt the file.
But I am not sure what problem you are facing to restore the backup. 

It could be that the text file is in CRLF (if you are using windows). Make sure to modify the file to make the line endings LF before encrypting and reuploading.

adithyapi4, 2024-12-19:
I edited the the text file in LF using Visual Studio and encrypted it. Still facing the same issue when uploading it

akash07k, 2025-01-22:
Sadly I too am facing the same issue with my JCOW 414 (R 2.57) @itsyourap  can you please reopen the issue if any fix is found or is in works?

itsyourap, 2025-01-23:
I won't reopen the issue since the problem is on Jio's side since even the original backup isn't getting restored.

itsyourap, 2025-01-24:
I checked on my router (JCOW414 running R2.57) and it doesn't have this issue.

@akash07k Are you following https://broadband.forum/threads/jiofiber-router-jco4032-settings-import-export.231300/post-1672959 guide to view the appropriate restore settings?

Please read the last line also which states:

> If you need to use restore functionality, the restore form has a hidden field called `thispage` which is set to `backupRestore.html` . Change that to `factoryDefault.html` .

Have you done this? Without it, you will get 401 Unauthorised error

akash07k, 2025-01-24:
Hi @itsyourap  Ah, I was actually following the instructions but was seeing the wrong hidden `thispage` input element and there it was already set to the `factoryDefault.html` therefore, I didn't bothered about it.
However, now with a closer look after reading your comment, I inspected the whole block very carefully and saw that there are multiple hidden inputs with `thispage` and one of them was correctly instead the `restore`.
For my case, I was looking at the `tf1_frmBackupFactoryDefaultSettings` block instead of `restore` block.
So, I correctly updated and now I have the root.
Thanks very much for the help and I must say, you are a life saver. Thanks my buddy.
❤️

Now I need to find the ways to enable bridging mode, USB etc. via root and many other things.
Way to begin the adventure.
Will be even more thankful if you can kindly point me to some important resources for enabling bridging etc.

> I checked on my router (JCOW414 running R2.57) and it doesn't have this issue.
>
> [@akash07k](https://github.com/akash07k) Are you following [...]

akash07k, 2025-01-24:
@itsyourap  couple of questions:
1. Does the root go away once the router reboots? In my case it is no more and everytime I am required to restore the modified file in order to get it back. is it normal behaviour?
2. Can we block the auto update of the firmware after getting root and if the point 1 is true, then can we make the root persistent somehow?
3. I think that I'll need to unpack the original firmware image in order to see all the file and directory structure. then only I'll be able to understand what can be modified, any specific way to reflash the firmware if I edit some files? will I need to sign the modified image somehow?
4. Is it possible to get out of double nat/cgnat with root?

PotatoPC44, 2025-01-24:
Use this to make root persist after every reboot and also there's a discussion on this github stating a way to disable and capture firmware updates.

https://yashgarg.dev/notes/jiofiber-bits

PotatoPC44, 2025-01-24:
This is link to block and capture updates. 

https://github.com/JFC-Group/JF-Customisation/discussions/67

adithyapi4, 2025-01-24:
It turns out it I did not follow the steps thoroughly as specified in the forum and I made the same mistake as @akash07k . Root access works perfectly now.
Thank you so much! @akash07k @itsyourap

itsyourap, 2025-01-24:
> [@itsyourap](https://github.com/itsyourap?rgh-link-date=2025-01-24T15%3A47%3A11.000Z) couple of questions:
>
> 1. Does the root go away once the router reboots? In my case it is no more and everytime I [...]

@akash07k 

1. #45 
2. #67 
3. You can use the `firmd`/`mtd` commands to directly flash to your router. 
	You can also use the web interface to upload your img file and a fake sig file but before clicking on the restore button, run [this script](https://gist.github.com/itsyourap/51ea164b311d4b9bc5beaa69d4db602b) as root on your router which actually forces the firmware signature check to be success. Btw, I don't recommend reflashing custom made stuffs since they might break your router but anyways, you are free to do whatever you want if you know what you are doing.
4. No idea. You can open a discussion on this repo and if other people know about it then they can help you.

closed 2025-01-24 by itsyourap

adithyapi4, 2025-01-24:
@akash07k In case you didn't know, a workaround for Jio's CGNAT is to use IPv6. Nowadays, most applications support IPv6.

akash07k, 2025-01-25:
@PotatoPC44  thanks buddy.
Actually I'm using this code from the guide in order to make the root persistent upon each root:
```
#!/bin/sh
 
# Wait for the system to be ready
sleep 10
 
# Remove the telnet disable flag
rm /flash/telnetDisable
rm /flash/telnetDisableACS
 
# Start telnet and allow it through the firewall
/usr/sbin/telnetd
/pfrm2.0/bin/iptables -I fwInBypass -p tcp --dport 23 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT
 
# Set the root password
echo -e "rilpass\nrilpass" | passwd root
```

However, when router starts and I try to telnet into it, I always get wrong password error which is really weird because if I manually execute this same script via shell, then it works and I can login with updated password.
Any ideas?
I have created the customInit file at `/flash2/pfrm2.0/etc/customInit`.
Actually there was no `pfrm2.0` directory inside flash2, so I have manually created it.

I also tried another approach:
```
cat > customInit << 'EOF'
#!/bin/sh

LOG_FILE="/tmp/customInit.log"
exec > "$LOG_FILE" 2>&1

sleep 10

rm /flash/telnetDisable || echo "Failed to remove telnetDisable flag"
rm /flash/telnetDisableACS || echo "Failed to remove telnetDisableACS flag"

/usr/sbin/telnetd || echo "Failed to start telnet"
/pfrm2.0/bin/iptables -I fwInBypass -p tcp --dport 23 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT || echo "Failed to configure iptables"

# Set the root password and log it
PASSWORD="rootpwd"
echo "root:$PASSWORD" | /usr/sbin/chpasswd || echo "Failed to set root password"
echo "Password for root set to: $PASSWORD"

EOF
chmod +x customInit

```
but same results.
Am I still missing something?
I also inspected `/etc/shadow` after restoring the modified .enc from the router admin panel and it doesn't contain any entry for root.
Any ideas where I'm wrong?
Would appreciate if you have other working way or script.
> Use this to make root persist after every reboot and also there's a discussion on this github stating a way to disable and capture firmware updates.
> 
> https://yashgarg.dev/notes/jiofiber-bits
> 
> This is link to block and capture updates.
> 
> [#67](https://github.com/JFC-Group/JF-Customisation/discussions/67)

akash07k, 2025-01-25:
Amazing insight bro.
Quite useful, I'll experiment with multiple things once I can make the root persistent.
Also, can I directly put c`customInit` or any other script inside `flash2` or it strictly needs to be in `flash2/pfrm2.0/etc` directory?

> For you other questions. You can experiment however you want with the root access and just a brief introduction about firmware. It's a embedded Linux firmware which reset it's file system on every boot so any changes you do won't be permanent that's why you are losing root access. From other people's research only flash2 is persistent storage which survives reboot and probably reset too (This is a major security flaw, you can easily write scripts to share all network traffic to your VPS even if you rest the router so you can easily snoop on new owner of this modem) so you need to write scripts in that and to make changes automatically after every boot. That's how the previous script i shared works.
> 
> please share your findings on discussion part of this github.
> 
> No you cannot disable cgnat because it's from the carrier's side, to disable this you will need to gain root access to Jio's backend. But there are many other ways to get around this for example using a home server to forward all traffic to a vps etc.
> 
> Also check and contribute discussions part of this github you will find very useful information there.

akash07k, 2025-01-25:
Wow, routing/sharing whole network traffic to the VPS or even raspberryPI etc is an interesting in order to see that what exactly is going on and experiment further.

> For you other questions. You can experiment however you want with the root access and just a brief introduction about firmware. It's a embedded Linux firmware which reset it's file system on every [...]

akash07k, 2025-01-25:
Are you able to make the root persistent upon reboot?

> It turns out it I did not follow the steps thoroughly as specified in the forum and I made the same mistake as [@akash07k](https://github.com/akash07k) . Root access works perfectly now. Thank you so much! [@akash07k](https://github.com/akash07k) [@itsyourap](https://github.com/itsyourap)

akash07k, 2025-01-25:
Ah, that will be great bro.
Will surely be experimenting with IP V6 then

> [@akash07k](https://github.com/akash07k) In case you didn't know, a workaround for Jio's CGNAT is to use IPv6. Nowadays, most applications support IPv6.

akash07k, 2025-01-25:
Wonderful.
Thanks a lot bro.
So, any specific format or filesystem/offset config in which the firmware needs to be repacked after modification?
I think I need to learn the firmd and mtd commands for sure.
However, I'll modify the firmware very carefully in order to avoid any major brick issues.

> > [@itsyourap](https://github.com/itsyourap?rgh-link-date=2025-01-24T15%3A47%3A11.000Z) couple of questions:
> > 
> > 1. Does the root go away once the router reboots? In my case it is no more and everytime I am required to restore the modified file in order to get it back. is it normal behaviour?
> > 2. Can we block the auto update of the firmware after getting root and if the point 1 is true, then can we make the root persistent somehow?
> > 3. I think that I'll need to unpack the original firmware image in order to see all the file and directory structure. then only I'll be able to understand what can be modified, any specific way to reflash the firmware if I edit some files? will I need to sign the modified image somehow?
> > 4. Is it possible to get out of double nat/cgnat with root?
> 
> [@akash07k](https://github.com/akash07k)
> 
> 1. [Run script in startup #45](https://github.com/JFC-Group/JF-Customisation/issues/45)
> 2. [Capture JF Firmware URL #67](https://github.com/JFC-Group/JF-Customisation/discussions/67)
> 3. You can use the `firmd`/`mtd` commands to directly flash to your router.
>    You can also use the web interface to upload your img file and a fake sig file but before clicking on the restore button, run [this script](https://gist.github.com/itsyourap/51ea164b311d4b9bc5beaa69d4db602b) as root on your router which actually forces the firmware signature check to be success. Btw, I don't recommend reflashing custom made stuffs since they might break your router but anyways, you are free to do whatever you want if you know what you are doing.
> 4. No idea. You can open a discussion on this repo and if other people know about it then they can help you.

adithyapi4, 2025-01-25:
> Are you able to make the root persistent upon reboot?

I have not tried it out yet. I will let you know as soon as I do so

PotatoPC44, 2025-01-25:
@akash07k 

Yes you have to manually create the directory in flash2 and give it 10-15 minutes after the boot to let the script run properly. It's normal to get wrong password error just after boot.

itsyourap, 2025-01-25:
@akash07k Yes! @PotatoPC44 is correct! After the router LED turns green, wait atleast 5 minutes for the script to execute.

akash07k, 2025-01-25:
ok
> I have not tried it out yet. I will let you know as soon as I do so

akash07k, 2025-01-25:
Oh, I see.
10-15 minutes? that's too huge. but sadly Jio is Jio.
Ok, I'll wait for 15 minutes now before trying to login to telnet. just rebootted the router.

And, perfect, I'll update all of the things via customInit script only.
Now I need to search that how can I enable the bridging via root along with updating the Wifi SSID and password.
Please let me know if you have any pointers.

> [@akash07k](https://github.com/akash07k)
> 
> Yes you have to manually create the directory in flash2 and give it 10-15 minutes after the boot to let the script run properly. It's normal to get wrong password error just after boot.
> 
> And yes you need to make exact file in exact same directory manually for it to work. If you want to make any additional changes in the file system your only option is via this exact script. Or you can make additional scripts anywhere in the persistent store but you will need to call that script from custominit script.

akash07k, 2025-01-25:
Ok, sounds great bro.
Thanks so much.
Really love all the help. once I'm done experimenting, I'll try to make a good wiki for documenting each of the minor things and FAQs and send a PR to this repo so that it becomes even easier for the people who are beginning their adventures with the ONT.

> [@akash07k](https://github.com/akash07k) Yes! [@PotatoPC44](https://github.com/PotatoPC44) is correct! After the router LED turns green, wait atleast 5 minutes for the script to execute.

itsyourap, 2025-01-25:
> So, any specific format or filesystem/offset config in which the firmware needs to be repacked after modification?

@akash07k Check the `environment` table in `/tmp/system.db`. It contains some informations about which mtd partition is used for what purpose and so on. Also check the `firmware.upgrade` function in `/pfrm2.0/share/lua/5.1/teamf1lualib/firmware.lua` to get an idea about what arguments you need to pass to `firmd` for modifying the firmware.

akash07k, 2025-01-25:
Perfect, ok bro.
thanks so much, I'm going to check it soon.

> [@akash07k](https://github.com/akash07k) Check the `environment` table in `/tmp/system.db`. It contains some informations about which mtd partition is used for what purpose and so on. Also check the [...]

itsyourap, 2025-01-25:
> Now I need to search that how can I enable the bridging via root along with updating the Wifi SSID and password. Please let me know if you have any pointers.

@akash07k Take a look at `/pfrm2.0/share/lua/5.1/teamf1lualib/easyMeshMgmt.lua` to get an idea about how the Wi-Fi SSIDs and passwords are configured. It uses `/userfs/bin/tcapi` to modify the settings.

```shell
/userfs/bin/tcapi set WLan_Entry0 SSID "MySSID"
/userfs/bin/tcapi set WLan_Entry0 WPAPSK "IamN00BHacker"
```

We have `WLan_Entry0`, `WLan_Entry1`,  `WLan_Entry2`,  `WLan_Entry3`, `WLan11ac_Entry0`, `WLan11ac_Entry1`, `WLan11ac_Entry2`, `WLan11ac_Entry3`, `WLan_Common`, `WLan11ac_Common`

Never really tested them but you are free to check.

adithyapi4, 2025-01-25:
> I have not tried it out yet. I will let you know as soon as I do so

@akash07k Root works after reboot

itsyourap, 2025-01-25:
Let's not continue any discussion here under this issue. Rather you can open a discussion for these.