JFC-Group/JF-Customisation #75: Help retrieving /flash/secure/key.txt from Jio Fiber Box JCO500

rbhorse, 2024-12-17:
I've followed the guide at https://github.com/JFC-Group/JF-Customisation/blob/main/Instructions/Get-Any-File-From-JF-ONT-Home-Gateway.md but I havent been able to figure out how to get the encryption key

Model: JCO500 (Square white box)
Firmware version:  FXCTF1_JCO500_R3.13
Serial: RUFOTCC000*****

Things I tried (and resulting error screenshot below):
Used Postman with configuration specified on the page
Used Zap Proxy, intercepted the request and edited to align with instructions given on the page

What might help:
A known-good cURL command where I can substiture the current token and TeamF1Login as a sanity check.
Information on what the USB label should look like; I've set the label for the FAT32 drive to `SSS` using Windows 10, should it appear with the same name under 'Administration'? Mine appears by the OEM name (I tried both names while sending the requests)- 
[image]

If all fails, any disassembly instructions? Does the box have hidden screws beneath the giant grey circular rubber pad or should I just pull on the top hat harder?

[image]

itsyourap, 2024-12-19:
Are you even sure your router uses the `/flash/secure/key.txt` for encrypting backup?
That was used almost one year ago and now new methods are used.

itsyourap, 2024-12-19:
Did you try to use the https://github.com/JFC-Group/JF-Customisation/blob/main/keyguesser.py ?

rbhorse, 2024-12-19:
> Did you try to use the https://github.com/JFC-Group/JF-Customisation/blob/main/keyguesser.py ?

This did not work.
I will probably just dump it from the flash chip in a couple of days

itsyourap, 2025-01-09:
@rbhorse Updates?

viraniac, 2025-01-12:
R3.13! interesting... I saw a post on onlytech forum where a person had R3.7 firmware on Jcow404 and from the screenshots that firmware didn't had the backupRestore or factoryDefault page. R3.13 sounds even newer than that. I wonder what more changes it has.

itsyourap, 2025-01-24:
Closing due to inactivity.

closed 2025-01-24 by itsyourap