JFC-Group/JF-Customisation #76: Help retrieving encryption key from JioFiber router

M-S-R-P, 2025-01-09:
I am trying to find the encryption key of my JioFiber router, here are the specs:
Model: **JCO4032**
Firmware Version: **SRCMTF1_JCO4032_R2.52.1**

I followed all the steps in the guide, but Postman is showing "Sending request..." and nothing happens.

itsyourap, 2025-01-09:
Use https://github.com/JFC-Group/JF-Customisation/blob/main/keyguesser.py

M-S-R-P, 2025-01-09:
Thank you, I got the key and obtained root access, but I can't figure out
how to install openwrt on it.

itsyourap, 2025-01-09:
You need to build OpenWRT on your own if you want to use it.

closed 2025-01-09 by itsyourap

Naitik1208, 2025-05-04:
> Thank you, I got the key and obtained root access, but I can't figure out
> how to install openwrt on it.
> […](#)

Hey can you tell how you obtained the key as keyguesser doesn't work anymore

M-S-R-P, 2025-05-04:
Hi, Turns out I didn't use the keyguesser in the right place in the file
structure, I used it after placing the file and the encrypted file in the
same folder and it worked

Naitik1208, 2025-05-04:
> file <br> structure, I used it after placing the file and the encrypted
> file in the <br> same folder and it worked <br> <br> On Sun, May 4, 2025,
> 07:16 Naitik1208 ***@***.***&gt; wrote: <br> <br> &gt; *Naitik1208* left a
> comment (JFC-Group/JF-Customisation#76) <br> &gt; &lt;
> <br> &gt; <br> &gt; Thank you, I got the key and obtained root access, but
> I can&#39;t figure out <br> &gt; how to install openwrt on it. <br> &gt; …
> &lt;#m_8184661543016058154_&gt; <br> &gt; <br> &gt; Hey can you tell how
> you obtained the key as keyguesser doesn&#39;t work <br> &gt; anymore <br>
> <br> &gt; &lt;

well the keyguesser worked and I also encrypted the file after disabling
tr069.
I found that you can't upload anything on router using wifi unlike other
routers bcz it will show ERR_CONNECTION RESET
so you have to use ethernet as well as uploading on ipv4 gateway (i mean
192.168.1.1 or 192.168.29.1) may not work but it works on
jiofiber.local.html very easily.

itsyourap, 2025-05-04:
Not sure but disabling tr069 on newer firmwares might cause a bootloop in the router.
Only fix is to reset.

Naitik1208, 2025-05-04:
Any ways to seprate ssids 2.4ghz and 5ghz and list more like its in JCW414_2.57 firmware  through config file in JCW404_3.10.2 firmware as in this firmware 2.4ghz is almost a trash 5ghz works fine but on 2.4ghz it takes about 20hrs to download 150MB even on 200Mbps connection

itsyourap, 2025-05-04:
@Naitik1208 Idk about it. I am currently on R2.57 and won't upgrade.