JFC-Group/JF-Customisation #78: sending request to backupRestore.html as form-data returns unauthorized 401 error while trying to copy files to a pendrive.

LeontKing2, 2025-01-26:
[image]
Yes, i've made sure to input the correct cookie and token.

LeontKing2, 2025-01-26:
Firmware  Ver:  SRCMTF1_JCOW414_R2.57

LeontKing2, 2025-01-26:
Model : JCOW414

LeontKing2, 2025-01-26:
just checked that all firmware revisions at 2.57 have hidden the form for backing up router settings and backupRestore.html yadayada.. probs gonna downgrade it or wait till new solution or firmware update..
It's pretty weird that jio hasn't released a new firmware update since July 22nd of last year.

itsyourap, 2025-01-26:
This method is old and not working now.

Use https://github.com/JFC-Group/JF-Customisation/blob/main/keyguesser.py

itsyourap, 2025-01-26:
Router Backup Restore Option can be un-hidden

Read https://broadband.forum/threads/jiofiber-router-jco4032-settings-import-export.231300/post-1672959

itsyourap, 2025-01-26:
Please look into the previously closed issues before raising a new one.

- Duplicate of #66 
- Duplicate of #76

closed 2025-01-26 by itsyourap

LeontKing2, 2025-01-27:
unfortunately, the keys don't work and at the end it just ends up using the routers serial number RSsomething for decryption and returns gibberish data. (keyguesser.py)

itsyourap, 2025-01-27:
Unexpected! I am myself using SRCMTF1_JCOW414_R2.57 and everything works great with keyguesser.

LeontKing2, 2025-01-27:
also, this router has been rooted before... i have rooted this but i assume that jio had somehow hard reset the config through tr-069 or through some firmware update. (Issue #44  )

LeontKing2, 2025-01-27:
it's also weird that jio only allows you to retrieve R2.35, R.2.39, R2.45 and R2.41 from the distribution URL.

itsyourap, 2025-01-27:
> also, this router has been rooted before... i have rooted this but i assume that jio had somehow hard reset the config through tr-069 or through some firmware update. (Issue [#44](https://github.com/JFC-Group/JF-Customisation/issues/44) )

That shouldn't cause any issue.

> it's also weird that jio only allows you to retrieve R2.35, R.2.39, R2.45 and R2.41 from the distribution URL.

J** changed their firmware URL because they know we are tracking it. 
Btw, the firmware URLs of SRCMTF1_JCOW414_R2.57 can be found [here](https://onlytech.com/community/threads/jiofiber-ont-firmware-updates.55330/post-618983)

LeontKing2, 2025-01-27:
I've just checked klitelib.so in a disassembler according to a previous pull request https://github.com/JFC-Group/JF-Customisation/pull/41 
and no new keystrings have been added.
so definitely not a key string issue.

LeontKing2, 2025-01-27:
I've asked claudeai to make the script bruteforce the thing and it worked.
```
#!/usr/bin/env python3

import subprocess
import tempfile
import os
from itertools import permutations
import random
import string
from typing import List, Optional, Generator, Tuple
from dataclasses import dataclass
import logging
from pathlib import Path
from datetime import datetime
import binascii
import time

@dataclass
class RouterConfig:
    serial: str
    ssid: str
    input_file: Path
    output_file: Path
    validation_words: List[str] = None

    def __post_init__(self):
        if self.validation_words is None:
            self.validation_words = ["config", "router", "password", "ssid", "wifi"]

# Configuration
ROUTER = RouterConfig(
    serial="RSXXXXXXXXXXXXXXXX",
    ssid="XXXXX",
    input_file=Path("RSXXXXXXXXXXXXXX_JCOW414.enc"),
    output_file=Path("RSXXXXXXXXXXXXXX_JCOW414.txt")
)

ENCRYPTION_KEYS = [
    "1n0NaZQnC9oxcfwf", "us4AQiJAgbj0Fmxq", "NTqK8Ps5iFke8zrp", "bfqerloC15y79WQZ",
    "9gNzEbuDjtyT9Pyc", "uuphsZuO92AZW5GJ", "qdySWmmvYKdBcO53", "Q7ODauKsxUAUtbR7",
    "Kohgiem4joochei3", "6f1D27JyLm70GUUu", "zuFbKywMhJjVEhk3", "6uMrt5ricsD1ABDh",
    "iPjZ8bYm6s3uGYVf", "QGwaPHx2K1rNDTmL", "fJ7OeRF2TvqKdR30"
]

class DecryptionDebugger:
    def __init__(self, config: RouterConfig):
        self.config = config
        self.temp_dir = Path(tempfile.mkdtemp(prefix="jiofiber_decrypt_"))
        self.debug_log = self.temp_dir / "debug.log"
        self.results_file = self.temp_dir / "decryption_results.txt"
        self.binary_results = self.temp_dir / "binary_results"
        self.successful_decryptions = []
        self.attempts = 0
        self.start_time = time.time()
        self._setup_logging()

    def _setup_logging(self):
        logging.basicConfig(
            level=logging.DEBUG,
            format='%(asctime)s - %(levelname)s - %(message)s',
            handlers=[
                logging.FileHandler(self.debug_log),
                logging.StreamHandler()
            ]
        )
        self.logger = logging.getLogger(__name__)

    def _generate_random_key(self, length: int = 16) -> str:
        """Generate a random key of specified length."""
[... 97 lines cut ...]
            if is_valid:
                print(f"\nValid decryption found!")
                print(f"Key: {key}")
                print(f"Hex Key: {hex_key}")
                print("Content preview:")
                print(content[:500])
                return True
        return False

def main():
    debugger = DecryptionDebugger(ROUTER)
    try:
        debugger.try_all_approaches()
    except KeyboardInterrupt:
        print("\nDecryption process stopped by user.")
    finally:
        print(f"\nTemporary files located at: {debugger.temp_dir}")
        print(f"Total attempts: {debugger.attempts}")
        print(f"Total time: {time.time() - debugger.start_time:.1f} seconds")

if __name__ == "__main__":
    main()
```

[image]

LeontKing2, 2025-01-27:
fixed script that outputs the file to output_file instead of just letting it sit there in temp.
```
#!/usr/bin/env python3

import subprocess
import tempfile
import os
from itertools import permutations
import random
import string
from typing import List, Optional, Generator, Tuple
from dataclasses import dataclass
import logging
from pathlib import Path
from datetime import datetime
import time
import shutil

@dataclass
class RouterConfig:
    serial: str
    ssid: str
    input_file: Path
    output_file: Path
    validation_words: List[str] = None

    def __post_init__(self):
        if self.validation_words is None:
            self.validation_words = ["config"]  # Simplified to just "config"

# Configuration
ROUTER = RouterConfig(
    serial="RSXXXXXXXXXXXXXXXX",
    ssid="XXXXX",
    input_file=Path("RSXXXXXXXXXXXXXX_JCOW414.enc"),
    output_file=Path("RSXXXXXXXXXXXXXX_JCOW414.txt")
)

class DecryptionDebugger:
    def __init__(self, config: RouterConfig):
        self.config = config
        self.temp_dir = Path(tempfile.mkdtemp(prefix="jiofiber_decrypt_"))
        self.debug_log = self.temp_dir / "debug.log"
        self.attempts = 0
        self.start_time = time.time()
        self._setup_logging()

    def _setup_logging(self):
        logging.basicConfig(
            level=logging.DEBUG,
            format='%(asctime)s - %(levelname)s - %(message)s',
            handlers=[
                logging.FileHandler(self.debug_log),
                logging.StreamHandler()
            ]
        )
        self.logger = logging.getLogger(__name__)

    def _generate_random_key(self, length: int = 16) -> str:
        charset = string.ascii_letters + string.digits + string.punctuation
        return ''.join(random.choice(charset) for _ in range(length))

    def _validate_and_save_content(self, content: str, key: str, hex_key: str) -> bool:
        """Validate content and save to output file if valid."""
        if any(word.lower() in content.lower() for word in self.config.validation_words):
            print(f"\nValid config found!")
            print(f"Key: {key}")
            print(f"Hex Key: {hex_key}")
            
            # Save to output file
[... 93 lines cut ...]
        """Clean up temporary files."""
        try:
            if self.temp_dir.exists():
                shutil.rmtree(self.temp_dir)
        except Exception as e:
            self.logger.error(f"Cleanup error: {str(e)}")

def main():
    debugger = DecryptionDebugger(ROUTER)
    try:
        if not debugger.config.input_file.exists():
            print(f"Input file not found: {debugger.config.input_file}")
            return
        
        debugger.try_decryption()
    except KeyboardInterrupt:
        print("\nDecryption process stopped by user.")
    finally:
        print(f"\nTotal attempts: {debugger.attempts}")
        print(f"Total time: {time.time() - debugger.start_time:.1f} seconds")
        debugger.cleanup()

if __name__ == "__main__":
    main()
```

LeontKing2, 2025-01-27:
the original ver of the keyguesser didn't want to go through everything for some reason. it just stuck to the serial key. i've just found that out.

LeontKing2, 2025-01-27:
@itsyourap don't we need to change the checksum of the config at the last after modifying it? it isn't present in this guide
https://github.com/JFC-Group/JF-Customisation/blob/main/Instructions/Alternate-Way-To-Get-Root-Access-JF-ONT-Home-Gateway.md

itsyourap, 2025-01-27:
> don't we need to change the checksum of the config at the last after modifying it?

Nope! Its not required if you are trying to get root access. Basically, if the checksum is correct, the router will try to replace the routers current config with your supplied one and then restart the router which we don't want.

We want the router to go through the supplied config file (which triggers the RCE) and then reject the config as checksum error. This way your router doesn't reboot and hence you get root access.

LeontKing2, 2025-01-27:
well, it is rejecting the config file due to the checksum error, but no port for telnet opened.

[image]

[image]

itsyourap, 2025-01-27:
Contact me on Telegram: @itsyourap

LeontKing2, 2025-01-27:
will contact when i retrieve access to les mobile phone.

itsyourap, 2025-02-26:
@SpeedxThunder001 This method is quite old. Don't use this. Probably it doesn't work anymore.

Use keyguesser

SpeedxThunder001, 2025-02-26:
yes i used keyguesser and succesfully edited and encrypted the key but i got checksum error when uploading the backup

SpeedxThunder001, 2025-02-26:
so i was able to restore it succesfully still 23 is blocked