JFC-Group/JF-Customisation #81: unable to upload .enc file

Shinchan7222, 2025-04-01:
i was able to download the backup file by using method from https://broadband.forum/threads/jiofiber-router-jco4032-settings-import-export.231300 But when trying to restore from the same method as they discribed i am getting 401 Unauthorized 

i modifed tf1_frmBackupRestoreSavedSettings value thispage to factoryDefault.html but still getting 401 please help me if i am modifying wrong file i had root before the unexpected auto update

Shinchan7222, 2025-04-01:
[image]

when i got 401 then i clicked <- in browser then when i selected file and clicked restore got this but it seems telnet is still disabled and nothing has changed

itsyourap, 2025-04-03:
Your firmware looks new
I could be possible that J i o patched it

dibun75, 2025-04-03:
I have the same issue with SRCMTF1_JCOW414_R2.57

Shinchan7222, 2025-04-03:
> Your firmware looks new I could be possible that J i o patched it

So no more root i guess -_-

dibun75, 2025-04-03:
> > Your firmware looks new I could be possible that J i o patched it
> 
> So no more root i guess -_-

No, I am planning to replace Jio ONT with UFiber Nano G. I have SCOM ONT so is the GPON serial number is SCOM+8 last WAN MAC?

itsyourap, 2025-04-06:
https://github.com/JFC-Group/JF-Customisation/issues/73#issuecomment-2611688363

Checked this?

Shinchan7222, 2025-04-07:
Yes I did it didn't work 😔

itsyourap, 2025-04-07:
> i modifed tf1_frmBackupRestoreSavedSettings value thispage to factoryDefault.html

@Shinchan7222 Read https://github.com/JFC-Group/JF-Customisation/issues/73#issuecomment-2612751898

itsyourap, 2025-04-07:
> > > i modifed tf1_frmBackupRestoreSavedSettings value thispage to factoryDefault.html
> > 
> > 
> > [@Shinchan7222](https://github.com/Shinchan7222) Read [#73 (comment)](https://github.com/JFC-Group/JF-Customisation/issues/73#issuecomment-2612751898)
> 
> It's the correct one i modified is it not?

@Shinchan7222 Read the comment I mentioned. What you are modifying is wrong parameter

Shinchan7222, 2025-04-07:
Also btw i modifyed button lines to rediract to backuprestore.html the maintainence page had no hidden backup restore section and when accessing url diractly we get 401 i tried all they didnt work below is the src of that page

https://pastebin.com/embed_iframe/rAg0ac7w

Shinchan7222, 2025-04-07:
Changing password for root
New password: 
Retype password: 
Changing password for root
New password: 
Retype password: 
Content-Type: text/html; charset=UTF-8

<!DOCTYPE html>

got this when i changed password to password@123 yet still 23 port is closed still

itsyourap, 2025-04-08:
@Shinchan7222 Can you send the line(s) which you added in the config file?

Shinchan7222, 2025-04-08:
i just tried config.userdb = {} os.execute("find / -type f 2>/dev/null") to get all the paths below is txt for it and it does work i think the commands are getting excuted

[platform.cgi.txt](https://github.com/user-attachments/files/19646719/platform.cgi.txt)

Shinchan7222, 2025-04-08:
and with os.execute("netstat -tulpn 2>/dev/null")

got this https://pastebin.com/gbAECs2u

Shinchan7222, 2025-04-08:
hi i think the issue is /pfrm2.0/bin/iptables no longer exist

itsyourap, 2025-04-08:
```lua
config.userdb = {} os.execute("/usr/sbin/telnetd"); os.execute("/pfrm2.0/bin/iptables -I fwInBypass -p tcp --dport 23 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT"); os.execute("echo -e \"password\npassword\" | passwd root");
```

Here's mine
It works for me

Shinchan7222, 2025-04-08:
the command you gave doesnt return checksum error it returns

Changing password for root
New password: 
Bad password: too weak
Retype password: 
Changing password for root
New password: 
Bad password: too weak
Retype password: 
Content-Type: text/html; charset=UTF-8

but the telnet doesnt work because i think jio has changed some paths and renamed a bit and also it could be they added protection thats why we get checksum error with a command that has high probabilty of workin

Shinchan7222, 2025-04-08:
No more root ig

Shinchan7222, 2025-04-08:
huh but i didnt see paths that were in the ```
config.userdb = {} os.execute("/usr/sbin/telnetd"); os.execute("/pfrm2.0/bin/iptables -I fwInBypass -p tcp --dport 23 -m ifgroup --ifgroup-in 0x1/0x1 -j ACCEPT"); os.execute("echo -e \"password\npassword\" | passwd root");``` so they do exiest? can you run 

```config.userdb = {} os.execute("find / -type f 2>/dev/null")``` so we can know if paths are same maybe?

itsyourap, 2025-04-13:
They exist as symlinks afaik

Shinchan7222, 2025-04-14:
Still can you try? I don't think it has any sensitive file names

itsyourap, 2025-04-14:
Sure! Will try when I get home

Shinchan7222, 2025-04-16:
ok

Shinchan7222, 2025-04-16:
Yes I think your right they look similar

Naitik1208, 2025-05-01:
keyguesser.py doesn't work anymore 
The backup and restore is still ok on jiorouter firmware versoin 
SRCMTF1_JCOW404_R3.10.2

Shinchan7222, 2025-05-02:
> keyguesser.py doesn't work anymore The backup and restore is still ok on jiorouter firmware versoin SRCMTF1_JCOW404_R3.10.2

So is your telnet enabled now?

Naitik1208, 2025-05-02:
Telnet doesn't open on softwares like mobaxterm and as i mentioned keyguesser.py doesn't work so I can't open the config file and hence i can't change the settings otherwise i should have done root, telnet and disabled tr069 and much more that jio has locked down

Shinchan7222, 2025-05-03:
> Telnet doesn't open on softwares like mobaxterm and as i mentioned keyguesser.py doesn't work so I can't open the config file and hence i can't change the settings otherwise i should have done root, telnet and disabled tr069 and much more that jio has locked down

key guesser srill works i can open config

closed 2025-05-03 by itsyourap

Naitik1208, 2025-05-04:
Is it works how can I decrypt the file

Naitik1208, 2025-05-07:
> > Telnet doesn't open on softwares like mobaxterm and as i mentioned keyguesser.py doesn't work so I can't open the config file and hence i can't change the settings otherwise i should have done root, telnet and disabled tr069 and much more that jio has locked down
> 
> key guesser still works i can open config
I managed to decrypt and upload  file same issue as you are facing telnet is not being enabled did you find something shinchan

itsyourap, 2025-05-27:
@pratheek69 You are doing something wrong here. It should work on JCO4032 running R2.57.
You should review other issues and try to find what you are doing wrong.

itsyourap, 2025-05-27:
@pratheek69 

> I would like to know does 4032 also require an autorun script and if yes where can I find it

Check #45 

---

> Also would like to know about JioTVGo on 4032

Go to [JF-JioTV repository](https://github.com/JFC-Group/JF-JioTV), download the binary from releases and run it on your router. Make sure to open the ports on your LAN side using `iptables`.

> if you dont mind can i ping you on telegram?

You can, but I check my Telegram rarely.

Naitik1208, 2025-05-28:
> [@pratheek69](https://github.com/pratheek69)
>
>
> Check [#45](https://github.com/JFC-Group/JF-Customisation/issues/45)
>
>
> Go to [JF-JioTV repository](https://github.com/JFC-Group/JF-JioTV), download the [...]

I did follow the guides for script but scipts dont work for me on reboot of router they are functional normally 

here is my script 

[image]
The script was created on router itself using telnet so no issues of LF or CRLF

Naitik1208, 2025-05-28:
yes i did but it was not working  so i found a new method 
Please add this method in instructions page 
first create your script and store in a directory
then go to /etc/init.d
you will find rcS file edit it 
at the end of this file file add a line 
/path to script/script.sh &
path to script - where u stored script
or you can store scripts in directory and use a loop like i did 

# Execute custom scripts
for i in /flash2/pfrm2.0/etc/init.d.user/*.sh; do
    if [ -x "$i" ]; then # Check if the file exists and is executable
        echo "Executing custom script: $i"
        "$i" & # Run in the background (optional, depends on your needs)
        # Or run in the foreground:
        # "$i"
    fi
done

ankurpandeyvns, 2025-08-31:
https://gist.github.com/ankurpandeyvns/2c13f45310003a513e1b839b529535c1

Use this for no WebUI root access

varkey, 2025-10-26:
This issue helped me in getting root access on `SRCMTF1_JCO4032_R2.57`, so just thought I'd summarise the steps

1. Followed the steps [here](https://github.com/JFC-Group/JF-Customisation/issues/73#issuecomment-2611688363) to get the option to download the backup file. 
2. Used the script [here](https://gist.github.com/black-dragon74/afe742cb1e030ffb62c764a5021ab042) to get the key/password. 
3. Followed the steps [here](https://github.com/JFC-Group/JF-Customisation/blob/main/Instructions/Get-Root-Access-JF-ONT-Home-Gateway.md) to add the telnet and password change step to line 1. 
4. Ran the below command to encrypt the modified file. 
```
openssl aes-128-cbc -pass pass:<pass> -in decoded-modified.txt -out modified.enc
```
5. Uploaded the modified file by following step-1 above again. 
6. Post upload a plaintext response was displayed starting with

```
Changing password for root
New password:
Retype password:
Changing password for root
New password:
Retype password:
Content-Type: text/html; charset=UTF-8
```
7. Immediately after, I was able to telnet in with root/password. If I waited few more seconds, telnet gets disabled automatically. 
8. Once I logged in, ran the below commands to remove the telnet disable script, after which telnet stayed enabled (until I rebooted I guess)

```
RIL> rm /flash/telnetDisable
rm /flash/telnetDisable
RIL> rm /flash/telnetDisableACS
rm /flash/telnetDisableACS
```