JFC-Group/JF-Customisation #83: how to downgrade JioFiber firmware in root?

luminous128, 2025-04-13:
I am on SRCMTF1_JCOW414_R2.57 firmware and I wanted to downgrade the firmware. the R2.57 update has disabled majority of the features...
I have root access of the router but I don't know how to downgrade the firmware in root
also, is there any script to disable auto update?

itsyourap, 2025-04-13:
If you want to downgrade the firmware, you can use https://gist.github.com/itsyourap/51ea164b311d4b9bc5beaa69d4db602b
Run the script and try performing a downgrade from the WEB admin panel

closed 2025-04-14 by luminous128

PotatoPC44, 2025-05-22:
Hi! Where can i find Old firmware files for my JCOW414. I have root access and i'm on 2.57 currently and wants to downgrade but idk how to get old firmware files.