JFC-Group/JF-Customisation #87: Telnet Refused connection

Naitik1208, 2025-05-04:
Upon trying to open telnet it refuses connection 
on executing os.execute and follwing every step correctly after uploading config file result is 
Changing password for root
New password: 
Retype password: 
Changing password for root
New password: 
Retype password: 
Content-Type: text/html; charset=UTF-8
but still telnet is not enabled neither TR069 is disabled only the thing that changed on router settings is the time till admin can be logged into page 

USING MOBAXTERM FOR TELNET

Naitik1208, 2025-05-04:
How will be able to know if root is complete

itsyourap, 2025-05-04:
No idea about the issue you are facing.

Naitik1208, 2025-05-04:
Probably you should add a tag that TR069 and root doesn't work anymore

itsyourap, 2025-05-04:
[image]

Naitik1208, 2025-05-04:
Please help me out if you find something about telnet btw i have also enabled telnet in firewall but I don't know why it's not working.
Well your work in microsip it's fantastic

closed 2025-05-17 by itsyourap