JFC-Group/JF-Customisation #92: The updated method for root didn't work for me.

vaibhav423, 2025-06-12:
**Device:** JCO110  
**Firmware :** ARCTFI JCO110_R1.148

---

### 🔧 Issue Summary

The updated root access method described in [this guide](https://github.com/JFC-Group/JF-Customisation/blob/master/Instructions/Alternate-Way-To-Get-Root-Access-JF-ONT-Home-Gateway.md) did **not work** for me on the JCO110 device running firmware version **R1.148**.

---

### ✔️ What Worked for Me

To successfully gain root access, I had to:

1. **Downgrade the firmware to `1.130`**, using files from this fork:  
   ➤ [Gangadhararao29/JioFiber-Home-Gateway @ JCO110/1.130](https://github.com/Gangadhararao29/JioFiber-Home-Gateway/tree/master/Firmwares/JCO110/1.130)  
   *(The original repo no longer contains the necessary firmware and related files.)*

2. Thankfully, **downgrading did not brick** my device.

3. Once on `1.130`, I was able to extract the **`server.key`** directly from the firmware — no need for `/flash/secure/key.txt`.

4. The rest of the rooting procedure continued normally after this point.

---

### 🔍 Additional Findings

- On JCO110 (R1.148), the **`key.txt` file mentioned in the guide does not exist** — neither in `/flash/secure/` nor anywhere else.

#### Telnet Session Output:
```
$ telnet 192.168.29.1

Connected to 192.168.29.1
reliance.reliance login: root
Password:

BusyBox v1.17.4 (2020-06-15 11:07:25 IST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

RIL> find / -type f -name key.txt 2>/dev/null
RIL> find / -type f -name https.key 2>/dev/null
/var/certs/self/default/https.key
/data/https.key
```

I'm not sure if this missing `key.txt` behavior is specific to the JCO110 model or affects others as well.

---

Would you please consider **re-adding the firmware files  and  keys** to the main repo, as they were before?  

It took me a good while to realize the required files were in a fork — having them available here , would **save time and effort for others** following the same guide. 😅

Thanks for your great work maintaining this project!

Gangadhararao29, 2025-06-14:
you can use the router_firmware.js file in the latest repo to get the firmwares, anyway jio has updated url for the latest versions and they are not available there

vaibhav423, 2025-06-14:
@Gangadhararao29 ,Thanks .  Now http://fota.slv.fxd.jiophone.net/ seems to work . it wasnt responding earlier , so I thought the server was shutdown.

now i could get the fw for JCO110 but couldn't for JCO4032 😕 .

vaibhav423, 2025-06-14:
@Gangadhararao29 , where u able to restore config . since in newer version the backup/restore is hidden in souce . however I was able to modify the source and could see both option but for example when I download a backup file and restore the same , I couldn't restore it . it shows Unauthorized login . ( I could confirm that no token / cookie were expired by going back again . )

itsyourap, 2025-06-15:
> now i could get the fw for JCO110 but couldn't for JCO4032 😕 .

@vaibhav423 even we couldn't. JCOW414 seems to have similar firmware as JCO4032.

closed 2025-06-15 by itsyourap