JFC-Group/JF-Customisation #95: DHCP not working

Naitik1208, 2025-06-15:
Suddenly after performing a reset DHCP server on my router has stopped working and password of admin page is changed from Jio centrum to something else, old password is also not working.

I can access root from telnet.
So is there any method to restore DHCP from telnet or manually flash firmware image file from telnet like we do in admin panel.

itsyourap, 2025-06-15:
This is the sole reason I don't recommend people changing stuff on a writable root partition, not even the `rcS` file.
I am not sure what broke on your end so I cannot help regarding this.

itsyourap, 2025-06-15:
To flash the firmware image, the `firmd` command can be used. We don't have any guidelines for its usage tho since it varies from router to router.

itsyourap, 2025-06-15:
Also, in a JCO4032 and JCOW414, we have the `mtd` command to flash our partitions with a new squashFS image. But since your router does not run on squashFS, I am not sure how to flash anything.

Naitik1208, 2025-06-15:
It was a problem from acs server (TR-069) their servers were not sending config files after reset for router to work

closed 2025-06-15 by Naitik1208